mpostol/ASMD #52: Dependencies are unavailable

ccvca, 2019-11-14:
Hello,
The CAS-Dependencies are not included in this github repository. Is this intended?
https://github.com/mpostol/ASMD/blob/5659c2b7cc2d4b54a454f5842988f32156217297/ModelDesigner.DesignStudio/ModelDesigner.DesignStudio.csproj#L94-L141

Some of them can be found on your github Account, but not all of them.

Greetings

ccvca, 2019-11-14:
After some reloads, Visual Studio (and I) found the CAS-Dependencies.
After unloading the Test-Project there are only some Errors missing:
That the following Namespaces are not available:
```
using DocumentFormat.OpenXml;
using DocumentFormat.OpenXml.Packaging;
using DocumentFormat.OpenXml.Wordprocessing;
```

ccvca, 2019-11-14:
Found a solution myself:
DocumentFormat.OpenXml, Version=2.5.5631.0
This version is no longer available, but installing 2.7. compiles fine.

mpostol, 2019-11-14:
Thanks, I will review the documentation.  It is related to #36 and #50 so fix it as a slave.

mpostol, 2019-11-14:
Duplicates #4.

closed 2019-11-15 by mpostol