mpostol/ASMD #87: UA.ModelDesigner.exe is not code-signed

JimLuth, 2019-12-18:
UA.ModelDesigner.exe is not code-signed.  Unsigned executables will not install on my computer (by company IT policy).

mpostol, 2019-12-18:
@JimLuth - I am aware of this. For now, the main goal is to publish the tool as an open-source for developers that can build this code using Visual Studio. The binaries have been added just to make the adoption process shorter. I will add a note that the zipped content is not signed. The installer package containing signed stuff is on this project road map - see the attached issue above. The main purpose of this project is further research/prototyping prioritized by the sponsors/users. Now I am working on the sponsor program - to be honest, I am starting from scratch. Let me know if this way we could join our efforts to improve the OPC UA specification, compliance, adaptability, research, training, etc. To discuss details you may contact me by email mpostol (at) cas.eu.

Anyway, many thanks for the assistance.

mpostol, 2019-12-18:
I have added a note to the [release page](https://github.com/mpostol/ASMD/releases/tag/4.1.0).
@JimLuth  close the issue if it works for you.  I will appreciate your assistance.

mpostol, 2019-12-21:
Consider using [SignTool.exe [Sign Tool)](https://docs.microsoft.com/en-us/dotnet/framework/tools/signtool-exe).

mpostol, 2020-03-30:
@JimLuth 

Hi! I'm excited that you are interested in contributing to Object-Oriented Internet initiatives.

The project collecting several repositories is maintained under the umbrella of the [Object-Oriented Internet (OOI) Partnership Program](https://github.com/sponsors/mpostol). The requested functionality is available for the mentioned partnership program.

As a result of participation in the sponsorship program, direct contribution or using the deliverables as the end user you

- can directly influence further development of the OOI prioritizing the tasks backlog
- will access OOI documentation and deliverables
- may use the deliverables in commercial products
- may distribute the deliverables including source code
- will have opportunities to join working groups, online discussion groups, and collaborations for new technology initiatives
- will have access to and direct contribution to new publications
- will make use of the OOI logo
- will have the opportunity to list your products in the online product guide
- will have the opportunity to announce new solutions and products online

## Consider joining

Now the priority is to clean up the dependency hell - several NuGet packages must be hopefully ported to .NET Standard, released and published.

Regards,
Mariusz

closed 2020-03-30 by mpostol