mpostol/ASMD #106: clone error

mregen, 2020-03-27:
Tried to clone, but no success

```
Git failed with a fatal error.
remote error: upload-pack: not our ref fd858831ccf71f7f2ae42b93b1cf2b5374a6ee86
The remote end hung up unexpectedly
Fetched in submodule path 'CommonResources', but it did not contain fd858831ccf71f7f2ae42b93b1cf2b5374a6ee86. Direct fetching of that commit failed.
```

mpostol, 2020-03-28:
Dependency on `CommonResources` submodule must be removed.

mpostol, 2020-03-28:
I have removed dependency on the submodule `CommonResources`.
Unit Test :+1:
Hope it works now.

mregen, 2020-03-30:
I can clone now, but the ModelingProject is tagged incompatible and I can't open it. So build is failing for the whole project. Maybe this is the reason?  --> https://devblogs.microsoft.com/devops/uml-designers-have-been-removed-layer-designer-now-supports-live-architectural-analysis/

mpostol, 2020-03-30:
In master, I always keep the error-free solution and tested using existing Unit Tests. What VS are you using? 
Anyway, I will remove this project. You are right it is obsolete.

mpostol, 2020-03-30:
@mregen I have just removed the obsolete projects. Unfortunately, they shall not be recognized as a source of compilation errors. Sometimes the compilation errors are caused by the late loading of the NuGet packages by VS. Check the References folder to make sure that all assemblies are visible. 

The project is maintained under the umbrella of the [sponsorship program](https://github.com/sponsors/mpostol). Now the priority is to clean up the dependency hell - several NuGet packages must be hopefully ported to .NET Standard, released and published.

Let me know how it works for you.

mpostol, 2020-05-11:
Hi @mregen,
I assume it works for you.  I'm excited that you are interested in contributing to Object-Oriented Internet initiatives.

The project collecting several repositories is maintained under the umbrella of the [Object-Oriented Internet (OOI) Partnership Program](https://github.com/sponsors/mpostol).

Consider joining. 
Mariusz

closed 2020-05-11 by mpostol