mpostol/ASMD #177: ObjectsFolder does not show other components than the server.

fgolra, 2020-09-11:
I have defined a folder object in my information model and imported the nodeset2 of this model in the ASMD v4.3.  If I organize this folder under the objectsfolder, the tool does not show this folder under "objects" in the browse view panel.

mpostol, 2020-09-11:
@fgolra can you send me the model created and the imported ones.

mpostol, 2020-09-12:
It contributes to #158

fgolra, 2020-09-14:
Please find attached, the concerned files.  You can import the nodeset to reproduce the bug.  
[Files.zip](https://github.com/mpostol/ASMD/files/5218187/Files.zip)

mpostol, 2021-02-03:
@fgolra I returned to work on Information Models. You will be available shortly to discuss your issue.
Is it still valid?